gfroerli/hardware #4: Add battery meter

dbrgn, 2017-08-15:
A battery meter would be really useful to track the battery voltage. (Maybe a regular ADC would be enough, but I think there are specialized chips for that.)

@rnestler do you think this should go into the first prototype? Do you have experience with these things?

rnestler, 2017-08-17:
> @rnestler do you think this should go into the first prototype? Do you have experience with these things?

I don't really have experience with this things. Also we have a Lio battery with a charging circuit and everything included, so maybe it can even provide this information?

An easy thing to do is just log the voltage of the battery.

dbrgn, 2017-08-17:
> with a charging circuit and everything included

It's only a protection circuit, it doesn't do battery level measuring. (There's no data wire attached :slightly_smiling_face:)

> An easy thing to do is just log the voltage of the battery, although the value of that is pretty low since Lio batteries have a flat charge/voltage curve.

Yes, that would be possible. Our microcontroller probably has a built-in ADC, right?

On the other hand, getting a percentage/charge value would be pretty nice. What about these that provide battery voltage / charge / temperature monitoring?

- [STC3100IST](https://www.digikey.com/product-detail/en/stmicroelectronics/STC3100IST/497-10526-1-ND/2270946) (1uA in power down, 100uA when operating)
- [TI bq27441-G1](https://www.digikey.com/product-detail/en/texas-instruments/BQ27441DRZR-G1A/296-39941-1-ND/5177818) (9uA in hibernate mode, 93uA when operating)
- [LC709203F](https://www.digikey.com/product-detail/en/on-semiconductor/LC709203FQH-01TWG/LC709203FQH-01TWGOSCT-ND/5801645) (3uA when operating)

If I understand it correctly, these chips must be always on, so 100uA / 93uA is way too much in our case, right? Or can they also be operated in sleep mode?

What about the third option?

rnestler, 2017-08-17:
>  LC709203F (3uA when operating)

I'm not sure if we can handle that one, it's [MSL](https://en.wikipedia.org/wiki/Moisture_sensitivity_level) 3 (168 Hours). So we can't keep it for longer than 168 hours before soldering...

dbrgn, 2017-08-17:
I see. But that's fine if we keep it in the original packaging until when we solder the boards, right?

rnestler, 2017-08-17:
>If I understand it correctly, these chips must be always on, so 100uA / 93uA is way too much in our case, right? Or can they also be operated in sleep mode?

I think you understand correctly. The main problem is, that those IC's are designed for smart-phones and other portable devices, which have a rather high current usage and get charged quite often.

dbrgn, 2017-08-17:
Hm, let's go with voltage measurements for now to keep it simple. That should be easy with the ADC, right?

The built-in 10 bit ADC supports the voltage range 0V-VDD. Since we get 3.3V from the LDO, we'd need voltage dividing resistors, correct?

rnestler, 2017-08-17:
> The built-in 10 bit ADC supports the voltage range 0V-VDD. Since we get 3.3V from the LDO, we'd need voltage dividing resistors, correct?

Correct!

dbrgn, 2017-08-22:
Hm, the voltage divider also uses up current. @rnestler can you ask at work whether they have experience with this kind of problem?

dbrgn, 2017-08-22:
Here's a potential solution with a transistor and a P-channel MOSFET: https://electronics.stackexchange.com/questions/64490/low-current-battery-monitoring#64491

[image]

What about that?

rnestler, 2017-08-23:
Isn't it possible to do it simpler? Just use an N-MOSFET to connect the voltage divider to GND? Or is the ADC input leakage to big?

dbrgn, 2017-08-23:
There are two comments on that answer regarding that:

> @jme - the ADC and MCU is referenced to ground and so it makes sense to switch the higher voltage feed. If I used an N channel device there would still be a drain permanently through the top resistor and thru the parasitic diodes in the MCU when it is in sleep mode.

> "Why not use a low side switch (e.g. N-ch FET or μC i/o pin)?" is a good question. Here's why. The battery voltage can be greater than Vcc. When the low side switch is opened, then the battery voltage will appear on the A/D pin. That could lead to the burnage of the A/D, or to a leakage of the battery through the protection diodes on the A/D pin.

I don't fully understand them though :)

rnestler, 2017-08-23:
Ah of course, silly me. If we disable the voltage divider the ADC input will be at battery voltage, which may or may not damage it, but certainly lead to a current flowing through the ESD diode to VDD.

dbrgn, 2017-08-23:
Ok, the schematic up there is a bit complex, but probably worth it. What do you think @rnestler? This should reduce the power requirements to the leak current of the FET and the BJT.

dbrgn, 2017-08-25:
What about this one?

https://www.digikey.ch/products/de?keywords=MAX6430EGUS%2BT-ND

It's a very simple dual-level voltage monitoring chip. This one (MAX6430EGUS) is factory-fixed to signal low_H at 3.3V and low_L at 3.0V and uses active push-pull for signal indication. (Open drain variants also exist but aren't available on Digikey.) It could be used to send a warning message / add a warning flag below 3.3V and turn off the circuit at 3.0V. (There are also user-configurable versions, but those would require external resistors.)

If you look at the voltage curve of a typical high capacity 18650 cell, it looks like this:

[image]

With low current consumption, at 3.3V, 3000 mAh have already been consumed. At 3.0V, additional 250 mAh have been consumed.

I think this could be a good alternative to the voltage divider, since it only requires a single part, no external components, and also turns off the entire circuit below a certain voltage. This would allow us to use unprotected Li-Ion cells, which are shorter and cheaper. Price is quite high though, at 4.40 CHF per piece when buying 10.

Downside when not having a voltage divider is that we don't know the voltage at all. We could also do both.

dbrgn, 2017-08-25:
Something like this, right @rnestler?

[image]

dbrgn, 2017-10-19:
Ping @rnestler. Maybe @ubruhin also wants to take a look?

ubruhin, 2017-10-20:
I guess your circuit does not work as expected. The voltage level of `LBOL` is referenced to the negative battery pin, but the voltage at the gate pin of the mosfet needs to be referenced to GND (which is *not* equal to the negative battery pin, it's floating as long as the mosfet is not yet switched on).

dbrgn, 2017-10-23:
We'll use the voltage divider and a DMC2038LVT-7DICT-ND dual MOSFET.

dbrgn, 2017-11-06:
Does this look good to you, @rnestler @ubruhin?

[image]

[Simulation at 5.2V](http://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+13.097415321081861+56+5+50%0Ar+464+160+464+224+0+6040%0Ar+464+272+464+336+0+9300%0Av+96+288+96+208+0+0+40+5.2+0+0+0.5%0Aw+96+288+96+336+0%0Aw+96+160+160+160+3%0Aw+96+160+96+208+0%0Af+208+288+256+288+0+1.5+0.02%0Af+352+224+352+160+9+1.5+0.02%0Aw+464+160+368+160+2%0Ar+256+224+256+176+0+10000%0Ar+192+288+192+336+0+10000%0Aw+160+288+192+288+0%0Aw+192+288+208+288+0%0Aw+256+304+256+336+0%0Aw+256+176+256+160+0%0Aw+256+224+256+272+0%0Aw+352+224+256+224+0%0Aw+256+160+336+160+0%0Aw+96+336+192+336+0%0Aw+192+336+256+336+0%0Aw+256+336+464+336+0%0Ap+528+256+528+336+1+0%0Aw+528+336+464+336+0%0Aw+528+256+464+256+0%0Aw+464+224+464+256+0%0Aw+464+256+464+272+0%0Ap+32+208+32+288+1+0%0Aw+32+288+96+288+0%0Aw+32+208+96+208+0%0As+160+288+160+160+0+1+false%0Aw+160+160+256+160+0%0A):

Disabled: 104 nA

[image]

Enabled: 1.4 mA

[image]

[Simulation at 3.3V](http://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+13.097415321081861+56+5+50%0Ar+464+160+464+224+0+6040%0Ar+464+272+464+336+0+9300%0Av+96+288+96+208+0+0+40+3.3+0+0+0.5%0Aw+96+288+96+336+0%0Aw+96+160+160+160+3%0Aw+96+160+96+208+0%0Af+208+288+256+288+0+1.5+0.02%0Af+352+224+352+160+9+1.5+0.02%0Aw+464+160+368+160+2%0Ar+256+224+256+176+0+10000%0Ar+192+288+192+336+0+10000%0Aw+160+288+192+288+0%0Aw+192+288+208+288+0%0Aw+256+304+256+336+0%0Aw+256+176+256+160+0%0Aw+256+224+256+272+0%0Aw+352+224+256+224+0%0Aw+256+160+336+160+0%0Aw+96+336+192+336+0%0Aw+192+336+256+336+0%0Aw+256+336+464+336+0%0Ap+528+256+528+336+1+0%0Aw+528+336+464+336+0%0Aw+528+256+464+256+0%0Aw+464+224+464+256+0%0Aw+464+256+464+272+0%0Ap+32+208+32+288+1+0%0Aw+32+288+96+288+0%0Aw+32+208+96+208+0%0As+160+288+160+160+0+1+false%0Aw+160+160+256+160+0%0A)

Disabled: 66 nA

[image]

Enabled: 873 uA

[image]

Total resistance: 15.34 kOhm.

The ADC reference is Vdd and we have 3.3V output from the LDO, so below that voltage we won't get usable results anymore (since the reference drops too). But that will only happen at the very end of the battery lifetime.