gfroerli/hardware #7: RN2483 ICSP Protection

dbrgn, 2017-08-17:
> During High Voltage In-Circuit Serial Programming
> mode, the RESET pin is driven with high-voltage (9V),
> therefore protection may be necessary for sensitive
> devices

Do we need something like that?

dbrgn, 2017-10-19:
@rnestler ping :)

rnestler, 2017-10-23:
Maybe we could just add a solder bridge?

rnestler, 2017-10-23:
> The  normal  firmware  upgrade  method  is  through  the
internal  bootloader  of  the  module  via  the  UART. 
> http://ww1.microchip.com/downloads/en/DeviceDoc/50002346C.pdf

So this would be highly unlikely that we ever need it. So a solder bridge between RN_RST and the pull-up resistor would definitely be enough.

dbrgn, 2017-11-06:
A solder bridge is a bit annoying if we have to add it for every board.

If we ever need to use ICSP, we can simply desolder the pull-up, that'll result in the same thing.

closed 2017-11-06 by dbrgn