gfroerli/hardware #8: Programming the RN2483

dbrgn, 2017-08-17:
Seems that it would be possible to program the RN2483 directly?

https://www.thethingsnetwork.org/forum/t/microchip-released-rn2483-library-code/4439

Would that suffice for our use case? It wouldn't be mbed compatible anymore though.

rnestler, 2017-08-17:
> The documentation states loading your own firmware on the module invalidates the certification.

From the linked forum. Is this a problem for us?

dbrgn, 2017-08-17:
Nope, as long as we don't sell our sensors as LoRaWAN certified devices :)

dbrgn, 2017-08-17:
The question is rather whether we have some significant limitations that we don't want to accept. With an external MCU we're probably more flexible.

dbrgn, 2017-10-19:
@rnestler we should probably stick to the external MCU, right?

rnestler, 2017-10-23:
For the moment yes.

closed 2017-10-23 by rnestler