gfroerli/hardware #16: Add hardware version detection

rnestler, 2017-11-20:
We are the worst... we forgot to add hardware detection 😄

dbrgn, 2017-11-21:
Good point 😉 

How is this usually being done, with some kind of ID chip or ROM?

rnestler, 2017-11-22:
Way to complicated 😉 Just pull some pins to GND.

ubruhin, 2017-11-22:
Or two resistors as voltage divider connected to an ADC pin.

rnestler, 2017-11-22:
> Or two resistors as voltage divider connected to an ADC pin.

Then there must flow current, which is not nice for a low power device. When pulling to GND we can just disable the internal pull-ups after reading the version.

rnestler, 2017-11-22:
Here is an example of what I mean:
[image]

dbrgn, 2017-11-23:
Sure, why not :)