gfroerli/hardware #24: Re-evaluate MCU

dbrgn, 2018-08-21:
What if we'd switch to an STM32F1 or similar for better Rust support, @rnestler?

rnestler, 2018-08-22:
> What if we'd switch to an STM32F1 or similar for better Rust support, @rnestler?

Would probably be a lot easier :wink: On the other hand we could just implement the stuff we need in https://github.com/coredump-ch/lpc11uxx-hal-rs.

dbrgn, 2018-08-22:
Yeah, would also be interesting. But also quite tricky to do correctly. See the [Rx](https://japaric.github.io/stm32f103xx-hal/src/stm32f103xx_hal/serial.rs.html#199) implementation for example... I would never have thought of the sequential memory ordering barrier.

rnestler, 2019-05-14:
https://www.digikey.ch/product-detail/de/stmicroelectronics/STM32L071KBT6/497-18577-ND/6166986 could be a good candidate. There are already a few crates around which should support it: https://crates.io/search?q=stm32l0

dbrgn, 2019-06-14:
We're using the STM32L071KBT6.

closed 2019-06-14 by dbrgn